bb-qq/r8152 #482: Apollo Lake 6.2 Drivers not installing in DS918, DSM6.2.4

dnguyen800, 2025-01-20:
### Description of the problem

> I am unable to install the r8152-apollolake-2.19.2-1_6.2.spk file on my Synology NAS DS918+ which uses Apollo Lake CPU. I'm running DSM 6.2.4-25556 Update 8. When trying to manual install, I receive an error message: "This package is not supported on the platform of Synology NAS or is incompatible with the current DSM version."

### Description of your products

> Synology NAS DS918+
> Linux nas 4.4.59+ #25556 SMP PREEMPT Fri Aug 16 14:57:02 CST 2024 x86_64 GNU/Linux synology_geminilake_920+
> DSM version 6.2.4-25556 Update 8
> UGREEN USB C to Ethernet Adapter 2.5G (2023)](https://amzn.to/3QHSElc) (RTL8156BG/Type-C)

dnguyen800, 2025-01-29:
I ended up upgrading to 7.2.2-72806, but still run into the same error message "This package is not supported on the platform of Synology NAS or is incompatible with the current DSM version." when trying to manually install "r8152-apollolake-2.19.2-1_7.2.spk".

Running this command "sudo install -m 4755 -o root -D /var/packages/r8152/target/r8152/spk_su /opt/sbin/spk_su" via SSH states an error message "No such file or directory".

bb-qq, 2025-02-16:
> Linux nas 4.4.59+ #25556 SMP PREEMPT Fri Aug 16 14:57:02 CST 2024 x86_64 GNU/Linux synology_geminilake_920+

The output of `uname -a` you wrote here shows is DS920+ (geminilake).
Could you try the driver for geminilake?

dnguyen800, 2025-02-16:
You are correct, the Gemini Lake drivers installed. I was following this Synology article to confirm the CPU architecture but the article incorrectly states my DS918 is Apollo Lake.
https://kb.synology.com/en-global/DSM/tutorial/What_kind_of_CPU_does_my_NAS_have

closed 2025-02-16 by dnguyen800