bb-qq/r8152 #484: Ds220+ with 2xSabrent 2.5gbps ethernet adapters

Maxgent1977, 2025-01-29:
Good evening,
at the momenti i can't seem to use 2.5 bandwidth when copying files from Ds220+ 10GB ram upgraded nas  to my desktop Win 11 pc.
Ds has 2xSabrent 2.5gbps ethernet adapters:

[image]

Rest of setup is like that 👍 

Desktop with dual rj45 rtl8125b network adapters connetcted directly to a QNAP QSW-1105-5T.
Sabrent adapters are hooked to that switch too and all ports are working at 2.5g (green led lit).
When i copy files from nas to pc i get only 1Gb speed running iperf3
Any help would be great.
Thanks in advance

Swimmer71, 2025-02-02:
How did you manage to see 2 of the USB/LAN adapters? I also tried 2 of UGREEN 2.5G USB to LAN adapters with my DS718+, but I only got LAN3 with 2.5G and never saw LAN4 with the second one. Info-center shows me both USB devices under USB-device.
I wanted to make a bond out of them.

Maxgent1977, 2025-02-02:
> How did you manage to see 2 of the USB/LAN adapters? I also tried 2 of UGREEN 2.5G USB to LAN adapters with my DS718+, but I only got LAN3 with 2.5G and never saw LAN4 with the second one. Info-center shows me both USB devices under USB-device. I wanted to make a bond out of them.

Hi, it's stated that you can bypass the "2 of same model adatpters issue" with this:

[image]

Swimmer71, 2025-02-03:
OK, thanks. with this bash... command I could see LAN 4. Now I wanted to make a bond2 out of LAN3+4 (Bond1 is already with  LAN1+2 with 1G each). This was not possible, it always returned to LAN3+4 with now Bond2.

When editing those files manually:
[image]

then restarting network with:
 /etc/rc.network restart

I got Bond 2 for a very short time:

[image]

But after restart, LAN3+4 were back.

[image]

Any ideas how to set up Bond2 with 2x 2.5G adapters?

bb-qq, 2025-02-16:
No, a configuration file for auto-loading must be placed under `/lib/modules-load.d/`.

By the way, what is your motivation to configure BOND with two USB adapters?
Of course throughput will not be improved, and if stability problems arise, they will be at the USB level, so two devices will fail at the same time.

bb-qq, 2025-02-16:
You can configure BOND as usual, but if you do not, BOND information will not be restored at the next startup.

What is your motivation to configure BOND with two USB adapters?
Of course throughput will not be improved, and if stability problems arise, they will be at the USB level, so two devices will fail at the same time.

Swimmer71, 2025-02-16:
Hi, the issue was, that I could NOT create a bond2 with 2x USBGreen adapters (LAN3+4) via UI. When I create it via UI, sandbox appears for about 30s, then LAN3+4 appear again (and no bond2).
The intension is to increase throughput if 2 local 2,5G clients access at same time.

Additionally I have 2x1G bond1 with LAN1+2 for external access, which could also be backup if USB devices habe problems.

Maxgent1977, 2025-02-16:
Anyone could help my issue...? Thanks

Swimmer71, 2025-02-16:
So PC, NAS and Switch LAN IF has 2.5GB and still no 2.5GB bandwith during copying? What is the MB/s rate when you copy large files with some GB?
Did you try it already with a different PC?

Maxgent1977, 2025-02-16:
Trasfer rate measured by irpef3 is a bit under the gigabit (980 or so) i tried opening a sessions n between Nas and another pc via Lan and got the same results. Copying to Nas imeasuring full speed. Smb is using protocol 3

Swimmer71, 2025-02-18:
Hmmm, sorry but I have no idea why you don't have 2.5G

bb-qq, 2025-03-20:
@Maxgent1977 Are you still slow if you use only one adapter? Have you also checked this page?
https://github.com/bb-qq/r8152/wiki/Troubleshooting#performance-issues