bb-qq/r8152 #486: DS414 with 2xClub 3D CAC-1420 hangs driver

endorianix, 2025-02-08:
### Description of the problem

As part of testing the possibility of using 2x Club 3D CAC-1420 adapters with my old DS414 (just for fun).
The configuration works correctly with a single adapter (tested on both rear USB ports). However, when connecting the second adapter (same Club 3D CAC-1420) it isnt recognized as network interface.

I used:
sudo bash /var/packages/r8152/scripts/install-udev-rules
However, this does not produce the desired effect.

### Description of your products

DS414
Club 3D CAC-1420 (x2)

Linux Syn_DS 3.2.101 #42962 SMP Mon Aug 19 15:11:03 CST 2024 armv7l GNU/Linux synology_armadaxp_ds414
DSM 7.1.1-42962 Update 7

Driver used: r8152-armadaxp-2.17.1-2_7.1 (the last one with support for armadaxp)

### Output of ``dmesg`` command

[ 6468.403172] r8152 4-1:1.0: eth3: chip rev 14
[ 6468.407650] usb 4-1: USB disconnect, device number 122
[ 6468.509702] etxhci_hcd-161118 0000:01:00.0: xHCI etxhci_drop_endpoint called with disabled ep ba5348e0
[ 6468.519042] etxhci_hcd-161118 0000:01:00.0: xHCI etxhci_drop_endpoint called with disabled ep ba53490c
[ 6468.528417] etxhci_hcd-161118 0000:01:00.0: xHCI etxhci_drop_endpoint called with disabled ep ba534938
[ 6468.659926] usb 4-1: new SuperSpeed USB device number 123 using etxhci_hcd-161118
[ 6468.687411] drivers/usb/core/ethub.c (1935) Same device found. Change serial to 401000000 
[ 6468.819917] usb 4-1: reset SuperSpeed USB device number 123 using etxhci_hcd-161118
[ 6468.843527] usb 4-1: device firmware changed
[ 6468.847876] r8152 4-1:1.0: (unregistered net_device): get_registers -19
[ 6468.854603] r8152 4-1:1.0: (unregistered net_device): Get ether addr fail
[ 6468.861909] r8152 4-1:1.0: eth3: v2.17.1 (2023/06/13)
[ 6468.866994] r8152 4-1:1.0: eth3: This product is covered by one or more of the following patents:
[ 6468.867002] 		US6,570,884, US6,115,776, and US6,327,625.

[ 6468.882753] r8152 4-1:1.0: eth3: chip rev 14
[ 6468.887254] usb 4-1: USB disconnect, device number 123

### Output of ``lsusb`` command

|__usb1          1d6b:0002:0302 09  2.00  480MBit/s   0mA 1IF  (Linux 3.2.101 ehci_hcd Marvell Orion EHCI ehci_marvell.0) hub
|__usb2          1d6b:0002:0302 09  2.00  480MBit/s   0mA 1IF  (Linux 3.2.101 ehci_hcd Marvell Orion EHCI ehci_marvell.1) hub
|__usb3          1d6b:0002:0302 09  2.00  480MBit/s   0mA 1IF  (Linux 3.2.101 etxhci_hcd-161118 Etron xHCI Host Controller 0000:01:00.0) hub
|__usb4          1d6b:0003:0302 09  3.00 5000MBit/s   0mA 1IF  (Linux 3.2.101 etxhci_hcd-161118 Etron xHCI Host Controller 0000:01:00.0) hub
  |__4-2         0bda:8156:3104 00  3.20 5000MBit/s  64mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 401000001)
  |__4-1         0bda:8156:3104 00  3.20 5000MBit/s  0IFs (Realtek USB 10/100/1G/2.5G LAN 401000000)

### Output of ``ifconfig -a`` command

(no eth3)

eth2      Link encap:Ethernet  HWaddr A0:CE:C8:AF:C7:40  
          inet addr:192.168.0.248  Bcast:192.168.0.255  Mask:255.255.255.0
          inet6 addr: fe80::a2ce:c8ff:feaf:c740/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:3737 errors:0 dropped:0 overruns:0 frame:0
          TX packets:561 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:1041280 (1016.8 KiB)  TX bytes:65183 (63.6 KiB)

bb-qq, 2025-02-16:
> |__4-2 0bda:8156:3104 00 3.20 5000MBit/s 64mA 1IF (Realtek USB 10/100/1G/2.5G LAN 401000001)
> |__4-1 0bda:8156:3104 00 3.20 5000MBit/s 0IF (Realtek USB 10/100/1G/2.5G LAN 401000000)

From this output, it appears that the port is not getting enough power.
The USB-C2500 has a more power-efficient 8156B chip, which may be the reason it worked so well.

By the way what is your motivation for adding two NICs?

endorianix, 2025-02-16:
> From this output, it appears that the port is not getting enough power. The USB-C2500 has a more power-efficient 8156B chip, which may be the reason it worked so well.

I was able to change serial number of one CAC-1420 and it works - so it looks that on old kernel hot plug fix doesnt work (maybe because of old udev?) 

> By the way what is your motivation for adding two NICs?

Just for fun (but at the end I see some performance gains, notably with SMB multichannel - which by definition should not work on this device... but works)

bb-qq, 2025-02-16:
> so it looks that on old kernel hot plug fix doesnt work (maybe because of old udev?)

I don't know the details, which is a bit odd since there is no intervening class driver for this output of lsusb.

> Just for fun (but at the end I see some performance gains, notably with SMB multichannel - which by definition should not work on this device... but works)

Understand. If the goal is to increase throughput, RTL8157 would be an option, just to let you know.

bb-qq, 2025-06-08:
The issue has been resolved, so this task will be closed. Please feel free to re-open it if you have any questions.

closed 2025-06-08 by bb-qq