bb-qq/r8152 #487: DS420 not working

Alex9-9, 2025-02-10:
Hi, i have a DS420 running DSM 7.2.2-72806 Update 3. Connected to a router with built in 2.5gbe switch. 

I installed driver 2.19.2-1_7.2 but even tho it is running it wont display 3rd nic. I tried reinstalling it a couple of times but every time it didn't display the 3rd nic it settings. When in package center it is displayed normally. At the end i tried to install 2.18.1-1_7.2, just to see if it would work. No luck. I guess it just won't work with DS420. I used geminilake version, DS420 uses Intel J4025.

P.S. I reinstalled the driver (to update it) on my DS920 in the exact same way, and it works perfectly.

[image]
[image]

webbz-dev, 2025-02-11:
I can confirm that after updating to 7.3 last night, the external 2.5G USB LAN adapter (UGREEN is no longer working.
Reinstalling drivers was to no avail.

bb-qq, 2025-02-16:
If you follow the template and attach the `lsusb` or `demsg` output I will look into it.

webbz-dev, 2025-02-16:
Sure, here you go!

[image]

lemle-a, 2025-02-21:
Hi,
https://github.com/bb-qq/r8152/issues/487#issue-2843363492
Please check  in the "package center" the driver is running after a reboot.
Mine was stopped. After I stared manually the additional eth card can be used :)

webbz-dev, 2025-02-22:
Thanks a lot, that did it for me as well :)

bb-qq, 2025-03-23:
I think the issue is resolved, so I will close this thread. If you have any questions, feel free to re-open it.

closed 2025-03-23 by bb-qq