bb-qq/r8152 #490: add r8169 (Model CM648) support to udev rules

blshkv, 2025-02-13:
Hi, 

first of all thanks a lot for this project. I have purchased UGREEN 2.5G LAN adapter and it works great.
The only issue is that I have to load the driver manually, so I end up adding it to autoboot script.

Please add it to the UDEV rules. Here is the lsusb output:

```
0bda:8156:31f0 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 0F13000000)
```

Thank you.

blshkv, 2025-02-13:
hm... It looks like it is already in the list:
https://github.com/bb-qq/r8152/blob/master/50-usb-realtek-net.rules#L12
```
ATTR{idVendor}=="0bda", ATTR{idProduct}=="815[2,3,5,6,7]"
```
any idea why it doesn't load automatically?

bb-qq, 2025-02-16:
If the driver package is enabled in the Package Center, it should load automatically. Is that not working?

Also, this driver does not use udev by default; see this article to enable udev rules.
https://github.com/bb-qq/r8152/wiki/Troubleshooting#hot-plugging-does-not-work

blshkv, 2025-02-16:
oh, I didn't know that udev rules are not getting installed by default. Ok, I installed it and the problem is fixed.

Thanks!

blshkv, 2025-02-16:
PS. Please consider adding that note to the "Installation" section. It not really a "Troubleshooting".

closed 2025-02-16 by blshkv