bb-qq/r8152 #492: Driver Update Procedure

Martlni, 2025-02-17:
Is there any recommended driver Update Procedure for example from 2.17.1 to 2.19.2?

Many thanks for wonderful Driver.
Martini

dreimer1986, 2025-03-04:
DS420+ here, but should work on any NAS version. Whenever I update the driver I just install the new one without uninstalling the old. For the case that something fails, like the SSH command line input is needed once more, I have my other LAN ports on the same IP settings. Thus I can just plug the cable into the other port then to check what went wrong and then switch back easily. Worked fine for years now. Most updates dont even need some interefering and just updated fine.

Martlni, 2025-03-04:
Halo Daniel,
many thanks for your feedback.
If you install the new driver without uninstalling the old one is there the input of SSH command line normaly neccessary or not?
I have connected both LAN and USB devices on two different IP's. Installation over the LAN IP is no problem. But only the crazy SSH command line is pretty annoying...

dreimer1986, 2025-03-04:
As I said, most of the time I did not need to fix the update by SSH. But sometimes you do. Say big thx to Synology for their messy system interna which cause that fix to be needed