bb-qq/r8152 #493: DS218 failed to start package

angipp01, 2025-02-18:
### Description of the problem
I've tried releases from 2.19 to 2.15 and all fail to run.  

### Description of your products

> - the product name of your NAS DS218
>     - and output of ``uname -a`` command
>     - Linux Ippolito 4.4.302+ #72806 SMP Thu Sep 5 13:44:21 CST 2024 aarch64 GNU/Linux synology_rtd1296_ds218
>     - DSM version 7.2.2-72806 Update 3
> - the product name of the ethernet dongle  = UGREEN 2.5GBS  USB-A to RJ45

### Description of your environment
Connected to Gig TP-link (unmanaged) Switch using Cat 6 cable. 

### Output of ``dmesg`` command

````
Please view the attached file

[dmesg.txt](https://github.com/user-attachments/files/18850603/dmesg.txt)

````

### Output of ``lsusb`` command

````

|__usb2          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller xhci-hcd.2.auto) hub
|__usb3          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller xhci-hcd.2.auto) hub
  |__3-1         0bda:8156:31f4 00  3.20 5000MBit/s 256mA 2IFs (Realtek USB 10/100/1G/2.5G LAN 4F13000001)
|__usb4          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller xhci-hcd.5.auto) hub
|__usb5          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller xhci-hcd.5.auto) hub
|__usb6          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller xhci-hcd.8.auto) hub
|__usb7          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller xhci-hcd.8.auto) hub

````

### Output of ``ifconfig -a`` command

````
~$ ifconfig -a
eth0      Link encap:Ethernet  HWaddr 00:11:32:A8:4B:44
          inet addr:192.168.2.6  Bcast:192.168.2.255  Mask:255.255.255.0
          inet6 addr: fe80::211:32ff:fea8:4b44/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:255867 errors:0 dropped:0 overruns:0 frame:0
          TX packets:180177 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:64398902 (61.4 MiB)  TX bytes:126493747 (120.6 MiB)
          Interrupt:11

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:13992 errors:0 dropped:0 overruns:0 frame:0
          TX packets:13992 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:2012901 (1.9 MiB)  TX bytes:2012901 (1.9 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

````

angipp01, 2025-02-22:
[image]

silencer1979, 2025-02-22:
having the same problem on DS218+. 
/var/packages/r8152/var/log says:

insmod: ERROR: could not insert module /volume1/@appstore/r8152/r8152/r8152.ko: Operation not permitted
rmmod: ERROR: Module r8152 is not currently loaded

Creo2005, 2025-03-01:
> having the same problem on DS218+. /var/packages/r8152/var/log says:
> 
> insmod: ERROR: could not insert module /volume1/@appstore/r8152/r8152/r8152.ko: Operation not permitted rmmod: ERROR: Module r8152 is not currently loaded
We will just install the driver for ds218+. DSM 6.2.4-25556 Update 8
Everything worked from scratch, there were no errors during installation.
USB 2.5Gbit adapter https://aliexpress.com/item/1005007812625631.html

But this model does not have a link aggregation.

bb-qq, 2025-03-20:
@angipp01 Could you attach `start-stop-status.log` and `synopkg.log`?
https://github.com/bb-qq/r8152/wiki/Troubleshooting#how-to-investigate-if-the-driver-does-not-work-properly

The driver does not seem to be loaded at all. The command you executed during installation may have failed.
```
sudo install -m 4755 -o root -D /var/packages/r8152/target/r8152/spk_su /opt/sbin/spk_su
sudo chmod a+rx /opt /opt/sbin
```

MrMo0, 2025-05-10:
I have the same problem but it is working when installing the driver through ssh :

sudo wget https://github.com/bb-qq/r8152/releases/download/2.19.2-2/r8152-broadwellnk-2.19.2-2_7.2.spk
sudo synopkg install r8152-broadwellnk-2.19.2-2_7.2.spk
sudo synopkg start r8152