bb-qq/r8152 #496: [DS423+] Ugreen 2.5 Gbit USB Adapter only runs at 1 Gbit speed, please help

rcv11x, 2025-02-28:
### Description of the problem

I'm not very good at explaining things since I'm a bit of a novice at this, but I'm going to explain it as best as possible to give as much information as possible, 3 days ago I got a new DS423+ with 3 drives, I bought a 2.5Gbit USB adapter which is [UGREEN 2.5G USB 3.0 to RJ45 Ethernet Adapter, 2500 Mbps Network Card](https://www.amazon.es/dp/B0CWV2Q6HJ), I applied the patch from this repo with the file `r8152-geminilake-2.19.2-1_7.2.spk` and following the guide I installed it correctly since in the network configuration it says "2500 Mbps, Full Duplex" and I set the MTU to 9000, right now I only have the USB adapter connected to the NAS and from the NAS to the router, the 2.5 USB adapter is connected to the router to a Fritzbox WAN/LAN port which is 2.5Gbit, on the same router I have configured this port to be for LAN5 (2.5Gbit) instead of WAN as I show in the image, and then the other 1 Gbit cable goes from the NAS to another 1 Gbit port on the Fritzbox router, this is not very important but I also have another cable that goes from the router to my room directly to a 2.5Gbit Switch and there I have my gaming PC connected which has a 2.5 card and the proxmox server which has another 2.5 card, there the tests with iperf3 do give 2.5gbit, but this is not important because I have the NAS in the dining room and it is directly connected to the router

doing tests with iperf3 from a VM to the NAS it only gives me 113MB/s which is the 1 Gbit connection and not 2.5Gbit, I don't know what I'm doing wrong

Router Config port

[image]

Jellyfin server - iperf3 test

[image]

Synology ds423+ - iperf3 test

[image]

### Description of your products

> - Synology DS423+
>     - Linux storenas 4.4.302+ #72806 SMP Thu Sep 5 13:44:44 CST 2024 x86_64 GNU/Linux synology_geminilake_423+
>     - DSM 7.2.2-72806 Update 3
> - UGREEN 2.5G USB 3.0 to RJ45 Ethernet Adapter, 2500 Mbps Network Card

### Description of your environment

< usb 2.5 ugreen > -------- < DS423+ > --------------- < Fritzbox router WAN/LAN port>

My main PC has a 2.5Gbit card.
Router: Fritzbox 5590
Synology NAS is directly connected to the router via Ugreen USB adapter

### Output of ``dmesg`` command

````

[dmesg_log.txt](https://github.com/user-attachments/files/19026261/dmesg_log.txt)

````

### Output of ``lsusb`` command

````
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 6500XXXXXXXXXXX)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-2         0bda:8156:31f4 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 4F13000001)
````

### Output of ``ifconfig -a`` command

````
eth0      Link encap:Ethernet  HWaddr 90:09:D0:6C:75:95
          inet addr:192.168.96.22  Bcast:192.168.96.255  Mask:255.255.255.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:88 base 0xe000

eth1      Link encap:Ethernet  HWaddr 90:09:D0:6C:75:96
          inet addr:169.254.130.6  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:89 base 0xa000

eth2      Link encap:Ethernet  HWaddr 6C:1F:F7:1B:9C:38
          inet addr:192.168.96.20  Bcast:192.168.96.255  Mask:255.255.255.0
          inet6 addr: fe80::6e1f:f7ff:fe1b:9c38/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:828761 errors:0 dropped:0 overruns:0 frame:0
          TX packets:402852 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:1234078007 (1.1 GiB)  TX bytes:235872460 (224.9 MiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:2030 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2030 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:274341 (267.9 KiB)  TX bytes:274341 (267.9 KiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

indy913, 2025-03-03:
First of all, I can confirm the adapter you mentioned works just fine on a DS423+. From the information you have presented, that also seems to be the case for you.

The screenshot of the FRITZbox (FB) suggests this box has one 2.5Gb connection, which you have used to connect to your DS423+. Unless you also use a 2.5Gb capable switch there, the connection to the other room is therefore only 1Gb.

bb-qq, 2025-03-23:
I am concerned that eth0 and eth2 are assigned the same subnet.

If the subnets are the same, the Linux kernel determines which NIC will be used by ifmetric. This could lead to a situation where the NICs do not communicate as intended.

indy913, 2025-03-23:
Is that not what the service order of the default gateway (network > general > "default gateway" EDIT) is for? And if you do use multiple subnets, you may want to check advanced settings (network > general > ADVANCED SETTINGS) and make sure that multiple gateways has been selected. That should precent asymmetrical routing.

Multiple LANs in the same subnet has never been an issue for me, not even for SMB multichannel.

bb-qq, 2025-03-23:
Though I'm not sure, since I've never actually used it, the above setting is likely a route metric that determines the selected subnet.

You could try unplugging the 1Gbps cable just to be sure.

closed 2025-06-08 by bb-qq