bb-qq/r8152 #498: Anyone get an RS814 to work?

ericbengson, 2025-03-04:
Anyone get an RS814 to work?

dreimer1986, 2025-03-05:
Which DSM Version do you have installed?
Why do you try to install a absolutely and terribly outdated SPK?
Remove it completely, reboot and use these:

If you use DSM 6.2:
https://github.com/bb-qq/r8152/releases/download/2.17.1-2/r8152-armadaxp-2.17.1-2_6.2.spk

If you use DSM 7.1:
https://github.com/bb-qq/r8152/releases/download/2.17.1-2/r8152-armadaxp-2.17.1-2_7.1.spk

I use the 7.1 version without any problems on my 2x DS414 Backup NAS that I use to Backup my DS420+ to.
Not much performance gain due to the slow CPU, but it does provide at least a lil bit more than the builtin Gigabit LAN port.

ericbengson, 2025-03-05:
7.1... Oh I only saw that version for xp, ill try it now!  Thanks!

ericbengson, 2025-03-05:
Installed 2.17.1-2 7.1 doesn't show up in networks like my Synology 220+ does hmmm...

dreimer1986, 2025-03-05:
Does it show up in the Synology Web UI? If so, try to give it a manually set IP instead of DHCP. Be 100% sure that DNS and Gateway are set right AND the Subnet Mask likely is 255.255.255.0 for you as for most ppl out there.

ericbengson, 2025-03-05:
No, doesn't show up in the web UI.  Ill work on network settings!  Thank you!

dreimer1986, 2025-03-06:
And the driver does show as started in app manager? Of yes then I have no clue and hope for some logs explaining it.