bb-qq/r8152 #503: Speed state not showing ds916 5gb RTL8157

fabio1605, 2025-03-16:
### Description of the problem

DS916 - 5GB RTL8157 - 5GB

Not showing a speed state on network status or via SSH - just shows blank - not sure if its causing any issues but speeds are a little unstable - copying files at 300mb then suddently dropping to 10-20mb

Tried running a ethtool on ssh and it just states "speed unknown"

Tried multiple reboots etc of both devices - the pc is getting a 5gb speed showing just not the USB Dongle on the synology nas

Connected to a RTL8157 pc dongle over cat6a

PC shows 5GB connection - synology shows blank on 2.19.2 version

### Description of your products

> - DS916+
>    Linux nas 3.10.108 #42962 SMP Mon Aug 19 15:13:15 CST 2024 x86_64 GNU/Linux synology_braswell_916+
>     -7.1
> WAVLINK USB C Ethernet Adapter, 5Gbps USB3.0/USB C RJ45

### Description of your environment

> -ETHERNET Direct
WAVLINK USB C - Both sides 
CAT6

### Output of ``dmesg`` command

[desgm.txt](https://github.com/user-attachments/files/19274648/desgm.txt)

[desgm.txt](https://github.com/user-attachments/files/19274649/desgm.txt)

### Output of ``lsusb`` command

|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (xhci_hcd 0000:00:14.0) hub
  |__1-5         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 6500645DB5E41693)
|__usb2          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (xhci_hcd 0000:00:14.0) hub
|__usb3          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
|__usb4          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
  |__4-1         0bda:8157:3000 00  3.20 5000MBit/s 544mA 1IF  (Realtek USB 10/100/1G/2.5G/5G LAN 0003000001)

### Output of ``ifconfig -a`` command

````
admin@nas:~$ ifconfig -a
bond0     Link encap:Ethernet  HWaddr 00:11:32:5D:39:C3
          inet addr:192.168.0.56  Bcast:192.168.0.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MASTER MULTICAST  MTU:1500  Metric:1
          RX packets:15504 errors:0 dropped:0 overruns:0 frame:0
          TX packets:15378 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:5205120 (4.9 MiB)  TX bytes:7668848 (7.3 MiB)

eth0      Link encap:Ethernet  HWaddr 00:11:32:5D:39:C3
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:12645 errors:0 dropped:0 overruns:0 frame:0
          TX packets:10048 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:4394017 (4.1 MiB)  TX bytes:5626629 (5.3 MiB)

eth1      Link encap:Ethernet  HWaddr 00:11:32:5D:39:C4
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:2859 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5330 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:811103 (792.0 KiB)  TX bytes:2042219 (1.9 MiB)

eth2      Link encap:Ethernet  HWaddr 80:3F:5D:FD:B5:08
          inet addr:192.168.1.1  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::823f:5dff:fefd:b508/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:2481 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2296 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:233097 (227.6 KiB)  TX bytes:627548 (612.8 KiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:1967 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1967 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:309610 (302.3 KiB)  TX bytes:309610 (302.3 KiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

tun0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          inet addr:192.168.20.1  P-t-P:192.168.20.2  Mask:255.255.255.255
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:500
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

````

bb-qq, 2025-03-23:
It is a known problem that link speeds are not displayed correctly. It is a display only issue and does not affect the actual communication.

I will attempt to fix it after I get the device.

fabio1605, 2025-03-24:
Hi

Thanks i did notice after it was a logged bug - my issue is i cant tell if im hitting 2.5gbps or 5gbps negotiation. Not sure how else i can tell

if i can help with my device let me know if any SSH access can help you

fabio1605, 2025-03-24:
Just to add i installed openSpeed test aswell

My speeds are 1658 Download /2450MBPS upload

It would suggest to me im negotiating at 2.5Gbps

bb-qq, 2025-06-02:
The display issue where the link speed was not displayed correctly on the RTL8157 has been fixed in the latest release.
Could you check actual link speed?