bb-qq/r8152 #510: 2.5G Interface freezes up and eventually resets only to freeze again.

kirbyan, 2025-03-26:
### Description of the problem

Apple MacBook Pro on 10.0.0.113 with UGREEN CM648 USB to 2.5G Ethernet. 
DS420j on Synology internal interface on 10.0.0.5 with UGREEN CM648 USB to 2.5G Ethernet on 10.0.0.6
Installed as described in your readme.  
Able to access DSM web interface through 2.5G on 10.0.0.6

Test 1...
Able to connect to a share and browse with SMB to 2.5G on 10.0.0.6
Writing file to share is OK
Reading file causes 2.5G Ethernet interface to lock up:
No communication or ping response to 10.0.0.6. DSM Web I/F unresponsive on 10.0.0.6 
The built-in Synology 1G interface continues to work on 10.0.0.5
Wait maybe 10 minutes and the 2.5G comes back, but locks up on further attempts to read files. 

Test 2...
Telnet/SSH into Synology over 1G internal interface on 10.0.0.5
On Synology run iperf -s
On Mac, run -perf -c 10.0.0.6 (to the 2.5G interface) 

andrew@Andrews-MBP-2 ~ % iperf -c 10.0.0.6 

------------------------------------------------------------
Client connecting to 10.0.0.6, TCP port 5001
TCP window size:  128 KByte (default)
------------------------------------------------------------
[  1] local 10.0.0.113 port 50396 connected with 10.0.0.6 port 5001
[ ID] Interval       Transfer     Bandwidth
[  1] 0.00-10.00 sec  2.68 GBytes  2.30 Gbits/sec
  
Seems good.... let's reverse roles.

On Mac, run iperf -s
On Synology run iperf -c 10.0.0.113

iperf hangs and now 2.5G interface is unresponsive 

From Mac, connect to DSM web: 
- check interface status in Control Panel - it says it is connected
- check logs and there is nothing in there  

### Description of your products

Synology DS420j
DSM 7.2.2-72806 Update 3
UGREEN Model CM648, Part Number 25051

andrew@Synology:~$ uname -a
Linux Synology 4.4.302+ #72806 SMP Thu Sep 5 13:44:21 CST 2024 aarch64 GNU/Linux synology_rtd1296_ds420j

### Description of your environment

In normal use the Mac and Synology connect via a Unifi Cloud Gateway Max router, with 2.5G ports, however I have replicated the setup using a generic unmanaged switch and static ip addresses set on all the interfaces.  

### Output of ``dmesg`` command

[ 1979.093655] CPU: 2 PID: 0 Comm: swapper/2 Tainted: P           OE   4.4.302+ #72806
[ 1979.101492] Hardware name: Realtek_RTD1296 (DT)
[ 1979.106131] task: ffffffc03c4fb000 ti: ffffffc03c504000 task.ti: ffffffc03c504000
[ 1979.113797] PC is at dev_watchdog+0x234/0x24c
[ 1979.118259] LR is at dev_watchdog+0x234/0x24c
[ 1979.122717] pc : [<ffffffc0006eaab8>] lr : [<ffffffc0006eaab8>] pstate: 60000145
[ 1979.130287] sp : ffffffc03c507c50
[ 1979.133679] x29: ffffffc03c507c50 x28: 0000000000000100 
[ 1979.139139] x27: ffffffc03c504000 x26: 0000000000000005 
[ 1979.144598] x25: 0000000000000020 x24: 00000000ffffffff 
[ 1979.150056] x23: 0000000000000002 x22: ffffffc0009e4000 
[ 1979.155516] x21: ffffffc00099e000 x20: ffffffc036fa3000 
[ 1979.160976] x19: 0000000000000000 x18: 0000000000000006 
[ 1979.166435] x17: ffffffbdc065de20 x16: ffffffc018b02600 
[... 155 lines cut ...]

eth1      Link encap:Ethernet  HWaddr 6C:1F:F7:1F:6C:FC  
          inet addr:10.0.0.6  Bcast:10.0.0.255  Mask:255.255.255.0
          inet6 addr: fe80::6e1f:f7ff:fe1f:6cfc/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:12079368 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2350935 errors:18 dropped:16 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:18180263381 (16.9 GiB)  TX bytes:314629095 (300.0 MiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:15322 errors:0 dropped:0 overruns:0 frame:0
          TX packets:15322 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:3672800 (3.5 MiB)  TX bytes:3672800 (3.5 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

kirbyan, 2025-03-26:
start-stop-status.log:

```
+ echo ----------
----------
+ date
Wed Mar 26 22:05:03 GMT 2025
+ grep -E '^(version|create_time)' /var/packages/r8152/INFO
version="2.19.2-1"
create_time="20241207-00:17:32"
+ echo status
status
+ case $1 in
+ /sbin/lsmod
+ grep r8152
r8152                 267985  0 
usbcore               213662  11 uas,r8152,usblp,uhci_hcd,usb_storage,ehci_hcd,ehci_pci,xhci_plat_hcd,usbhid,xhci_hcd,xhci_pci
+ exit 0
+ echo ----------
----------
+ date
Wed Mar 26 22:05:12 GMT 2025
+ grep -E '^(version|create_time)' /var/packages/r8152/INFO
version="2.19.2-1"
create_time="20241207-00:17:32"
+ echo stop
stop
+ case $1 in
+ set_interfaces down
+ action=down
+ target_interface_name=
++ ls /sys/class/net
+ all_interface_names='eth0
eth1
lo
sit0'
+ '[' -n ']'
+ '[' '!' -e /sys/class/net/ ']'
+ for interface_name in ${all_interface_names}
+ [[ -n '' ]]
+ '[' '!' -e /sys/class/net/eth0/device/driver ']'
++ ls -ld /sys/class/net/eth0/device/driver
+ driver_location='lrwxrwxrwx 1 root root 0 Mar 26 22:05 /sys/class/net/eth0/device/driver -> ../../../bus/platform/drivers/r8169'
++ echo 'lrwxrwxrwx 1 root root 0 Mar 26 22:05 /sys/class/net/eth0/device/driver -> ../../../bus/platform/drivers/r8169'
++ grep r8152
+ '[' '!' -z '' ']'
+ for interface_name in ${all_interface_names}
+ [[ -n '' ]]
+ '[' '!' -e /sys/class/net/eth1/device/driver ']'
++ ls -ld /sys/class/net/eth1/device/driver
+ driver_location='lrwxrwxrwx 1 root root 0 Mar 26 21:42 /sys/class/net/eth1/device/driver -> ../../../../../../../../bus/usb/drivers/r8152'
++ echo 'lrwxrwxrwx 1 root root 0 Mar 26 21:42 /sys/class/net/eth1/device/driver -> ../../../../../../../../bus/usb/drivers/r8152'
++ grep r8152
+ '[' '!' -z 'lrwxrwxrwx 1 root root 0 Mar 26 21:42 /sys/class/net/eth1/device/driver -> ../../../../../../../../bus/usb/drivers/r8152' ']'
+ config_file=/etc/sysconfig/network-scripts/ifcfg-eth1
+ config_storage_location=/var/packages/r8152/etc/ifcfg-eth1
+ '[' -f /etc/sysconfig/network-scripts/ifcfg-eth1 ']'
+ '[' down = down ']'
+ cp /etc/sysconfig/network-scripts/ifcfg-eth1 /var/packages/r8152/etc/ifcfg-eth1
+ ifconfig eth1 down
+ set_ovs_interface eth1 down
+ interface_name=eth1
+ action=down
+ config_file_ovs=/etc/sysconfig/network-scripts/ifcfg-ovs_eth1
+ config_storage_location_ovs=/var/packages/r8152/etc/ifcfg-ovs_eth1
+ is_ovs_enable
+ '[' '!' -f /usr/syno/etc/synoovs/ovs_reg.conf ']'
+ return 1
+ '[' down = up ']'
+ '[' down = up ']'
[... 143 lines cut ...]
+ echo status
status
+ case $1 in
+ /sbin/lsmod
+ grep r8152
r8152                 267985  0 
usbcore               213662  11 uas,r8152,usblp,uhci_hcd,usb_storage,ehci_hcd,ehci_pci,xhci_plat_hcd,usbhid,xhci_hcd,xhci_pci
+ exit 0
+ echo ----------
----------
+ date
Wed Mar 26 22:05:27 GMT 2025
+ grep -E '^(version|create_time)' /var/packages/r8152/INFO
version="2.19.2-1"
create_time="20241207-00:17:32"
+ echo status
status
+ case $1 in
+ grep r8152
+ /sbin/lsmod
r8152                 267985  0 
usbcore               213662  11 uas,r8152,usblp,uhci_hcd,usb_storage,ehci_hcd,ehci_pci,xhci_plat_hcd,usbhid,xhci_hcd,xhci_pci
+ exit 0

```

yunhao-jiang, 2025-04-17:
I think we are having the same issue. When the NAS do a downlink (or say when you upload something from the mac to NAS), the port become unresponsive.