bb-qq/r8152 #522: 2.19.2-2 - DS215j - WP-UT5 - This package is not supported on the platform of DiskStation or is incompatible with the current DSM version.

plawer, 2025-04-17:
### Description of the problem

> Uploading r8152-armada37xx-2.19.2-2_7.1.spk to Package center. When I click next, it returns "This package is not supported on the platform of DiskStation or is incompatible with the current DSM version." 

### Description of your products

> -DS215j
>     - Linux ds215j 3.2.101 #42962 SMP Mon Aug 19 15:10:10 CST 2024 armv7l GNU/Linux synology_armada375_ds215j

>     - DSM 7.1.1-42962 Update 8
> - https://www.wisdpi.com/products/wisdpi-usb-3-2-5g-ethernet-adapter-wp-ut5-wired-lan-network-connection-for-mac-os-linux-windows-backward-compatible-on-5g-2-5g-1g-100mbps-ideal-for-gaming

### Description of your environment
N/a

### Output of ``dmesg`` command
````
[    0.000000] Booting Linux on physical CPU 0
[    0.000000] Linux version 3.2.101 (root@build4) (gcc version 8.5.0 (GCC) ) #42962 SMP Mon Aug 19 15:10:10 CST 2024
[    0.000000] CPU: ARMv7 Processor [414fc091] revision 1 (ARMv7), cr=10c53c7d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine: Marvell Armada-375 Board
[    0.000000] Using UBoot passing parameters structure
[    0.000000] Memory policy: ECC disabled, Data cache writealloc
[    0.000000] On node 0 totalpages: 131072
[    0.000000] free_area_init_node: node 0, pgdat 8051de80, node_mem_map 80589000
[    0.000000]   DMA zone: 1024 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 130048 pages, LIFO batch:31
[    0.000000] PERCPU: Embedded 7 pages/cpu @8098d000 s5440 r8192 d15040 u32768
[    0.000000] pcpu-alloc: s5440 r8192 d15040 u32768 alloc=8*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 130048
[    0.000000] Kernel command line: console=ttyS0,115200 ip=off initrd=0x8000000 root=/dev/md0 rw syno_hw_version=DS215jv10 hd_power_on_seq=2 ihd_num=2 netif_num=1 flash_size=8
[    0.000000] Synology Hardware Version: DS215jv10-j
[    0.000000] Internal HD num: 2
[    0.000000] Internal netif num: 1
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 512MB = 512MB total
[    0.000000] Memory: 510484k/510484k available, 13804k reserved, 0K highmem
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
[    0.000000]     vmalloc : 0xa0800000 - 0xf4000000   (1336 MB)
[    0.000000]     lowmem  : 0x80000000 - 0xa0000000   ( 512 MB)
[    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
[    0.000000]       .text : 0x80008000 - 0x804b8000   (4800 kB)
[    0.000000]       .init : 0x804b8000 - 0x804e7540   ( 190 kB)
[    0.000000]       .data : 0x804e8000 - 0x8051ea40   ( 219 kB)
[    0.000000]        .bss : 0x8051ea64 - 0x8058818c   ( 422 kB)
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS:528
[    0.000000] Initializing Armada-375 SoC Timers
[    0.000000] sched_clock: 32 bits at 25MHz, resolution 40ns, wraps every 171798ms
[    0.000165] Calibrating delay loop... 1594.16 BogoMIPS (lpj=7970816)
[    0.090039] pid_max: default: 32768 minimum: 301
[    0.090177] Security Framework initialized
[    0.090205] AppArmor: AppArmor initialized
[    0.090271] Mount-cache hash table entries: 512
[    0.091048] CPU: Testing write buffer coherency: ok
[    0.091192] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.091218] Calibrating local timer... 399.83MHz.
[    0.151051] CPU1: Booted secondary processor
[    0.190030] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.190092] Brought up 2 CPUs
[    0.190106] SMP: Total of 2 processors activated (3188.32 BogoMIPS).
[    0.190875] devtmpfs: initialized
[    0.200348] NET: Registered protocol family 16
[... 571 lines cut ...]
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:16436  Metric:1
          RX packets:7907 errors:0 dropped:0 overruns:0 frame:0
          TX packets:7907 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:1126666 (1.0 MiB)  TX bytes:1126666 (1.0 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

tailscale Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          inet addr:x.x.x.x  P-t-P:x.x.x.x  Mask:255.255.255.255
          inet6 addr: xxxx:::::xxxx/128 Scope:Global
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1280  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:500
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

````

bb-qq, 2025-06-07:
I close this issue. Please feel free to reopen if you still have any questions.

closed 2025-06-07 by bb-qq

cmrdove, 2025-08-03:
Thank you! This worked for me too.