bb-qq/r8152 #523: Extreme latency after speed test on DS920+ & 8157

yunhao-jiang, 2025-04-17:
### Description of the problem

The driver runs smoothly until it has a workload on RX. As you can see, The Download (which is TX on synology) works perfectly fine. However, when it starts Upload testing (which is RX on synology), the speed drops significantly and eventually the NAS will be unresponsive (i.e., very high ping)

[image]

After the speed test, the latency surges and the web panel is unresponsive.
(before speed test, the average latency is around 0.3ms)
[image]

From wireshark, it look like this: (a lot of RST)

[image]

I think it's not the issue of openSpeedTest as I also tested with SMBv3. The upload (RX on synology) is also having the same issue.

The only way to solve this issue is to either reboot or unplug the ethernet cable and re connect, and then the ping and everything will start to work as usual.

I also turned off the intrusion protection and made sure the DOS protection is turned off on Synology.

### Description of your products

> - the product name of your NAS model
>     - `Linux host-name 4.4.302+ #72806 SMP Thu Sep 5 13:44:44 CST 2024 x86_64 GNU/Linux synology_geminilake_920+`
>     - DSM 7.2.2-72806 Update 3
> - WisdPi 8157 5Gbps

### Description of your environment

> - NAS - Switch; Switch - Mac mini M4 10GbE
>     - Ethernet switch: Ubiquiti Enterprise 8 PoE
> - cable category: Cat 6

### Output of ``dmesg`` command

(After the speed test starts)
````
[ 2479.962949] audit: type=1325 audit(1744926986.934:367): table=filter family=2 entries=12
[ 2479.973611] audit: type=1325 audit(1744926986.945:368): table=filter family=2 entries=11
[ 2479.984209] audit: type=1325 audit(1744926986.955:369): table=filter family=10 entries=10
[ 2479.994272] audit: type=1325 audit(1744926986.965:370): table=filter family=2 entries=10
[ 2480.004110] audit: type=1325 audit(1744926986.975:371): table=filter family=2 entries=11
[ 2480.013968] audit: type=1325 audit(1744926986.985:372): table=filter family=10 entries=9
````

### Output of ``lsusb`` command

````
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 650022F05113F209)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-1         0bda:8157:3000 00  3.20 5000MBit/s 544mA 1IF  (WisdPi USB 5G Ethernet 000334C8D6B103DC)
````

### Output of ``ifconfig -a`` command

````
eth2      Link encap:Ethernet  HWaddr 34:C8:D6:B1:03:DC  
          inet addr:192.168.1.100  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::36c8:d6ff:feb1:3dc/64 Scope:Link
          inet6 addr: fd41:1c57:a96a:584f:36c8:d6ff:feb1:3dc/64 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:578060 errors:0 dropped:0 overruns:0 frame:0
          TX packets:4720187 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:345178418 (329.1 MiB)  TX bytes:6798949530 (6.3 GiB)
````

yunhao-jiang, 2025-06-04:
@bb-qq Yes, both port front and back are having issue

bb-qq, 2025-06-07:
- Does the same problem occur even if the link speed is fixed at 1 Gbps or 2.5 Gbps?
- Does the same problem occur even when connecting directly to a Mac without going through a switch?