bb-qq/r8152 #536: Ds918 wake on lan and start driver after restart

filiphavlin, 2025-05-25:
Hello

I have ds918+ all work but
1. Not work wake on lan with new USB net adapter
2. After restart synology driver is not automatic run and must manual start
Log

[   15.392242] ata1.00: ATA-10: WDC WD40EFRX-68N32N0, 82.00A82, max UDMA/133
[   15.399863] ata1.00: 7814037168 sectors, multi 16: LBA48 NCQ (depth 31/32), AA
[   15.407946] ata1.00: SN:     WD-WCC7K2DY45Z2
[   15.413325] ata1.00: configured for UDMA/133
[   15.418790] scsi 0:0:0:0: Direct-Access     WDC      WD40EFRX-68N32N0
    0A82 PQ: 0 ANSI: 5
[   15.429019] sd 0:0:0:0: [sdc] 7814037168 512-byte logical blocks: (4.00 TB/3.64 TiB)
[   15.437724] sd 0:0:0:0: [sdc] 4096-byte physical blocks
[   15.443766] sd 0:0:0:0: [sdc] Write Protect is off
[   15.449131] sd 0:0:0:0: [sdc] Mode Sense: 00 3a 00 00
[   15.449204] sd 0:0:0:0: [sdc] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[   15.493088]  sdc: sdc1 sdc2 sdc5
[   15.497277] sd 0:0:0:0: [sdc] Attached SCSI disk
[   23.783002] ata2: link is slow to respond, please be patient (ready=0)
[   25.732098] ata2: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
[   25.776184] ata2.00: ATA-10: WDC WD40EFRX-68N32N0, 82.00A82, max UDMA/133
[   25.783783] ata2.00: 7814037168 sectors, multi 16: LBA48 NCQ (depth 31/32), AA
[   25.791856] ata2.00: SN:     WD-WCC7K6EKUDUF
[   25.797227] ata2.00: configured for UDMA/133
[   25.802660] scsi 1:0:0:0: Direct-Access     WDC      WD40EFRX-68N32N0
    0A82 PQ: 0 ANSI: 5
[   25.812902] sd 1:0:0:0: [sdd] 7814037168 512-byte logical blocks: (4.00 TB/3.64 TiB)
[   25.813171] Change defer qc mode on external port for compatibility
[   25.813172] Change defer qc mode on external port for compatibility
[   25.813173] Change defer qc mode on external port for compatibility
[   25.824144] ahci 0000:01:00.0: AHCI 0001.0000 32 slots 4 ports 6 Gbps 0xf
impl SATA mode
[   25.824150] ahci 0000:01:00.0: flags: 64bit ncq sntf led only pmp fbs pio
slum part sxs
[   25.824820] scsi host2: ahci
[   25.825001] scsi host3: ahci
[   25.825166] scsi host4: ahci
[   25.825256] ata3: SATA max UDMA/133 abar m2048@0x91500000 port 0x91500100
irq 379
[   25.825259] ata4: SATA max UDMA/133 abar m2048@0x91500000 port 0x91500180
irq 379
[   25.825262] ata5: SATA max UDMA/133 abar m2048@0x91500000 port 0x91500200
irq 379
[   25.895464] sd 1:0:0:0: [sdd] 4096-byte physical blocks
[   25.901487] sd 1:0:0:0: [sdd] Write Protect is off
[   25.906851] sd 1:0:0:0: [sdd] Mode Sense: 00 3a 00 00
[   25.906910] sd 1:0:0:0: [sdd] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[   25.954160]  sdd: sdd1 sdd2 sdd5
[   25.958341] sd 1:0:0:0: [sdd] Attached SCSI disk
[   34.179471] ata3: link is slow to respond, please be patient (ready=0)
[   35.871554] ata3: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
[   35.915764] ata3.00: ATA-10: WDC WD40EFRX-68N32N0, 82.00A82, max UDMA/133
[   35.923365] ata3.00: 7814037168 sectors, multi 16: LBA48 NCQ (depth 31/32), AA
[   35.931441] ata3.00: SN:     WD-WCC7K3TA4THY
[   35.936816] ata3.00: configured for UDMA/133
[   35.942283] scsi 2:0:0:0: Direct-Access     WDC      WD40EFRX-68N32N0
    0A82 PQ: 0 ANSI: 5
[   35.952562] sd 2:0:0:0: [sda] 7814037168 512-byte logical blocks: (4.00 TB/3.64 TiB)
[   35.961240] sd 2:0:0:0: [sda] 4096-byte physical blocks
[   35.967215] sd 2:0:0:0: [sda] Write Protect is off
[   35.972592] sd 2:0:0:0: [sda] Mode Sense: 00 3a 00 00
[   35.972694] sd 2:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[   36.031428]  sda: sda1 sda2 sda5
[   36.035622] sd 2:0:0:0: [sda] Attached SCSI removable disk
[   44.309921] ata4: link is slow to respond, please be patient (ready=0)
[   46.053008] ata4: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
[   46.098614] ata4.00: ATA-10: WDC WD40EFRX-68N32N0, 82.00A82, max UDMA/133
[   46.106222] ata4.00: 7814037168 sectors, multi 16: LBA48 NCQ (depth 31/32), AA
[... 652 lines cut ...]
[ 1513.052561] iSCSI:iscsi_target.c:642:iscsit_del_np CORE[0] - Removed Network Portal: 192.168.1.42:3260 on iSCSI/TCP
[ 1518.279528] audit_printk_skb: 36 callbacks suppressed
[ 1518.285183] audit: type=1325 audit(1748185918.169:252): table=filter family=2 entries=19
[ 1518.306044] audit: type=1325 audit(1748185918.196:253): table=filter family=10 entries=15
[ 1518.327397] audit: type=1325 audit(1748185918.217:254): table=filter family=2 entries=4
[ 1518.343298] audit: type=1325 audit(1748185918.233:255): table=filter family=2 entries=8
[ 1518.358762] audit: type=1325 audit(1748185918.248:256): table=filter family=2 entries=9
[ 1518.374353] audit: type=1325 audit(1748185918.264:257): table=filter family=2 entries=10
[ 1518.389853] audit: type=1325 audit(1748185918.279:258): table=filter family=2 entries=11
[ 1518.405453] audit: type=1325 audit(1748185918.295:259): table=filter family=2 entries=12
[ 1518.420961] audit: type=1325 audit(1748185918.311:260): table=filter family=2 entries=13
[ 1518.436528] audit: type=1325 audit(1748185918.326:261): table=filter family=2 entries=14
[ 3127.439412] iSCSI:iscsi_target.c:544:iscsit_add_np CORE[0] - Added Network Portal: 192.168.1.110:3260 on iSCSI/TCP
[ 3156.460850] iSCSI:iscsi_target.c:642:iscsit_del_np CORE[0] - Removed Network Portal: 192.168.1.109:3260 on iSCSI/TCP
[ 3156.472867] iSCSI:iscsi_target.c:544:iscsit_add_np CORE[0] - Added Network Portal: 192.168.1.42:3260 on iSCSI/TCP
[ 3336.016983] iSCSI:iscsi_target_parameters.c:39:iscsi_login_rx_data rx_data returned 0, expecting 48.
[ 3336.027233] iSCSI_F:iscsi_target_login.c:1303:iscsi_target_login_sess_out
iSCSI Login negotiation failed - I[][192.168.1.117:58854], T[][192.168.1.42:3260]
[ 3348.535901] iSCSI:iscsi_target_parameters.c:39:iscsi_login_rx_data rx_data returned 0, expecting 48.
[ 3348.546322] iSCSI_F:iscsi_target_login.c:1303:iscsi_target_login_sess_out
iSCSI Login negotiation failed - I[][192.168.1.117:34652], T[][192.168.1.42:3260]
[ 3349.215050] iSCSI:iscsi_target_parameters.c:39:iscsi_login_rx_data rx_data returned 0, expecting 48.
[ 3349.225306] iSCSI_F:iscsi_target_login.c:1303:iscsi_target_login_sess_out
iSCSI Login negotiation failed - I[][192.168.1.117:36332], T[][192.168.1.42:3260]
admin@DS918:/$ ethtool -h

filiphavlin, 2025-06-01:
Any tips?

bb-qq, 2025-06-02:
Is it possible to connect directly to the PC communicating with iperf without going through a router?
Could you please paste the iperf log here?

filiphavlin, 2025-06-04:
> Is it possible to connect directly to the PC communicating with iperf without going through a router? Could you please paste the iperf log here?

Hello I try on pc and speed is 2.3gb in attachment. It's good. I don't understand when iperf3 run in Asus router have also 2.5 port is speed 1.5. Both have same Lan cable and use same switch. 

[image]

bb-qq, 2025-06-07:
Iperf requires CPU power. So I suspect that the RT-AXE7800 did not have sufficient performance.