bb-qq/r8152 #537: UGreen USB C 2.5gbe adapter showing at 1000mbps connection and transferring at 42MBps instead of 200+

Germz89, 2025-05-30:
I bought a USB 2.5gbe adapter for my DS920+, installed the drivers, and have connectivity to my NAS, but it shows it as only a 1000mbps link instead of a 2500mbps link like in this video (he installed a 5gbps dongle, i did a 2.5gbe) https://youtu.be/iMJ-Lhyq16s?si=aVCNKftgt4YRnmj8&t=638

I'm using Cat6 cable from all devices. my Nighthawk router has 2.5gbe ports and so does my PC. Not sure what I'm doing wrong.

This is the driver that i installed. https://github.com/bb-qq/r8152/releases/download/2.19.2-2/r8152-geminilake-2.19.2-2_7.2.spk

bb-qq, 2025-06-07:
Have you check this page?
https://github.com/bb-qq/r8152/wiki/Troubleshooting#reverse-the-orientation-of-the-type-c-connector

pvdochtend, 2025-06-18:
I have the same problem.

For me it's:   |__1-3         0bda:8156:31f4 00  2.10  480MBit/s 200mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 4F13000001)

So it seems USB 2.0 according to your troubleshooting Guide. The UGreen USB-C to USB-A Adapter is not sufficient?
I am going to check if I can find a different USB C to USB A converter... I hope that's the issue and not the Adapter itself.

bb-qq, 2025-06-21:
@pvdochtend , have you try "flip" the type-C connector?