bb-qq/r8152 #540: DS920+ not working with both WAVLINK [WL-NWU340G Rev.a1] and

mdstoll, 2025-06-01:
### Description of the problem

I am having problems getting (one of) my two 5G Realtek 8157 USB dongles to work on my Synology DS920+ NAS. I've followed all explained detailed stepps but no new network connection shows up under network interfaces.

### Description of your products

**NAS**
Synology DS920+
DSM 7.7.2-72806 Update 3

```
sh-4.4# uname-a
Linux host-name 4.4.302+ #72806 SMP Thu Sep 5 13:44:44 CST 2024 x86_64 GNU/Linux synology_geminilake_920+```

```
sh-4.4# inxi
CPU: quad core Intel Celeron J4125 (-MCP-) speed/min/max: 2001/800/2001 MHz
Kernel: 4.4.302+ x86_64 Up: 5h 42m Mem: 2696.8/19852.1 MiB (13.6%)
Storage: 1.39 TiB (3595.1% used) Procs: 444 Shell: sh inxi: 3.3.27
```

**Dongle 1**
WAVLINK 
WL-NWU340G Rev.a1
USB-C 5 Gigabit Ethernet Adapter

_Same device as mentioned in this video:_ 
https://www.youtube.com/watch?v=SzMVNtzrxCM

```
sh-4.4# lsusb
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F00047D2513F267)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-1         0bda:8157:3000 00  3.20 5000MBit/s  0IFs (Realtek USB 10/100/1G/2.5G/5G LAN 0003803F5DD0ABA9)```

```
sh-4.4# usb-devices
T:  Bus=02 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=5000 MxCh= 0
D:  Ver= 3.20 Cls=00(>ifc ) Sub=00 Prot=00 MxPS= 9 #Cfgs=  3
P:  Vendor=0bda ProdID=8157 Rev=30.00
S:  Manufacturer=Realtek
S:  Product=USB 10/100/1G/2.5G/5G LAN
S:  SerialNumber=X
C:  #Ifs= 0 Cfg#= 0 Atr= MxPwr=
cat: '/sys/bus/usb/devices/usb2/2-1/2-*:?.*/bInterfaceNumber': No such file or directory
cat: '/sys/bus/usb/devices/usb2/2-1/2-*:?.*/bAlternateSetting': No such file or directory
cat: '/sys/bus/usb/devices/usb2/2-1/2-*:?.*/bNumEndpoints': No such file or directory
cat: '/sys/bus/usb/devices/usb2/2-1/2-*:?.*/bInterfaceClass': No such file or directory
cat: '/sys/bus/usb/devices/usb2/2-1/2-*:?.*/bInterfaceSubClass': No such file or directory
cat: '/sys/bus/usb/devices/usb2/2-1/2-*:?.*/bInterfaceProtocol': No such file or directory
/usr/local/bin/usb-devices: line 79: printf: (none): invalid number
I:  If#= 0 Alt= 0 #EPs= 0 Cls=() Sub= Prot= Driver=
```

**Dongle 2**
UGREEN 5Gbps USB C LAN Adapter Ethernet Adapter CM848

```
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-1         0bda:8157:3000 00  2.10  480MBit/s  0IFs (Realtek USB 10/100/1G/2.5G/5G LAN 0F03000001)
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F00047D2513F267)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
```

```
T:  Bus=01 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  6 Spd=480 MxCh= 0
D:  Ver= 2.10 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  3
P:  Vendor=0bda ProdID=8157 Rev=30.00
S:  Manufacturer=Realtek
[... 164 lines cut ...]

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

tun0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          inet addr:10.8.0.1  P-t-P:10.8.0.2  Mask:255.255.255.255
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:500 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

### EXTRA

When stopping and starting the package, dmesg outputs the following

```[23218.074247] Module [r8152] is removed. 
[23218.078560] usbcore: deregistering interface driver r8152
[23223.025276] usb 1-1: USB device idVendor=0bda idProduct=8157 manufacturer=Realtek product=USB 10/100/1G/2.5G/5G LAN is prohibited!
[23223.038474] usbcore: registered new interface driver r8152```

mdstoll, 2025-06-03:
> Not the owner of this repo but I have the [same ethernet adapter](https://amzn.to/4mKMX48) and [USB-A adapter](https://amzn.to/3Hp7IlG) as you. I'm using a DS1520+, so the same year as your NAS, and things are working for me. DSM 7.2.x. I'm seeing 3.4Gbps with 5GbE link using the adapter.
> 
> A few ideas:
> 
> * What is your DSM version? Which asset from release 2.19.2-2 are you using? Looks like you have a gemini lake CPU, so make sure you download the gemini lake asset and the correct DSM version.
> * The UGREEN USB-C to USB-A adapter is orientation sensitive on the USB-C side. One orientation gave me 400Mbps with 1GbE link. I then flipped it over and the other orientation gave me 4.7Gbps with 5GbE link (using a desktop).
> 
> The "is prohibited" part doesn't sound great...
> 
> > `[23223.025276] usb 1-1: USB device idVendor=0bda idProduct=8157 manufacturer=Realtek product=USB 10/100/1G/2.5G/5G LAN is prohibited!`

Hey, thanks for your reply!

Great to hear that you have a similar setup with a geminilake processor-NAS. Let me reply to your ideas:

1. 
DSM Verion is `DSM 7.7.2-72806 Update 3`
The asset I'm using is the [r8152-geminilake-2.19.2-2_7.2.spk](https://github.com/bb-qq/r8152/releases/download/2.19.2-2/r8152-geminilake-2.19.2-2_7.2.spk)
2. 
So I actually have two adapters that I both tried and that didn't work, but why not give it a try turning around the UGREEN one. 

Turning around the connector helped:
```sh-4.4# lsusb
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F00047D2513F267)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-1         0bda:8157:3000 00  3.20 5000MBit/s  0IFs (Realtek USB 10/100/1G/2.5G/5G LAN [S/N])```

However:

```[147587.086644] usb 2-1: new SuperSpeed USB device number 4 using xhci_hcd
[147587.105891] usb 2-1: USB device idVendor=0bda idProduct=8157 manufacturer=Realtek product=USB 10/100/1G/2.5G/5G LAN is prohibited!
[147587.119127] usb 2-1: USB device idVendor=0bda idProduct=8157 manufacturer=Realtek product=USB 10/100/1G/2.5G/5G LAN is prohibited!
[147587.132378] usb 2-1: USB device idVendor=0bda idProduct=8157 manufacturer=Realtek product=USB 10/100/1G/2.5G/5G LAN is prohibited!
[147587.145603] usb 2-1: USB device idVendor=0bda idProduct=8157 manufacturer=Realtek product=USB 10/100/1G/2.5G/5G LAN is prohibited!
[147587.158832] usb 2-1: USB device idVendor=0bda idProduct=8157 manufacturer=Realtek product=USB 10/100/1G/2.5G/5G LAN is prohibited!
[147587.172061] usb 2-1: USB device idVendor=0bda idProduct=8157 manufacturer=Realtek product=USB 10/100/1G/2.5G/5G LAN is prohibited!
[147587.185284] usb 2-1: USB device idVendor=0bda idProduct=8157 manufacturer=Realtek product=USB 10/100/1G/2.5G/5G LAN is prohibited!
[147587.198512] usb 2-1: USB device idVendor=0bda idProduct=8157 manufacturer=Realtek product=USB 10/100/1G/2.5G/5G LAN is prohibited!
[147587.211748] usb 2-1: USB device idVendor=0bda idProduct=8157 manufacturer=Realtek product=USB 10/100/1G/2.5G/5G LAN is prohibited!```

<img width="817" alt="Image" src="https://github.com/user-attachments/assets/7e575a2f-006b-4c37-b14a-5f458a03de6b" />

The prohibited error remains the same. 

Really unsure what to do with this or how to troubleshoot it. Searching for this 'lan prohibited' error doesn't resolve in any research-direction. Not with an AI search or with regular old Google.

mdstoll, 2025-06-03:
**Alright after some hard digging I figured it out!**

Apparantly and I'm not sure why, but the files 

`[50-usb-realtek-net.rules](https://github.com/bb-qq/r8152/blob/master/50-usb-realtek-net.rules)`
and 
`[51-usb-r8152-net.rules](https://github.com/bb-qq/r8152/blob/master/51-usb-r8152-net.rules)`

Weren't successfully created in (or copied to) directory `/lib/udev/rules.d/`

Just as with other linux systems using usb.conf, the device needs ot know that certain USB accessories are allowed. By means of these files, DSM knows this. Before it simply didn't allow the connection. After manually creating these files, the adapter now works.

For the love of me don't ask why this doesn't work natively on my system though...

bb-qq, 2025-06-03:
While I provide a mechanism to install udev.rules, I don't recommend it because it modifies the system. This is touched upon here:
https://github.com/bb-qq/r8152/wiki/Troubleshooting#hot-plugging-does-not-work

I don't understand why the driver wouldn't work without udev.rules being installed. Attaching these logs to this issue might help us understand what's happening:
https://github.com/bb-qq/r8152/wiki/Troubleshooting#how-to-investigate-if-the-driver-does-not-work-properly

bb-qq, 2025-06-07:
Yes

bb-qq, 2025-06-21:
Let me ask you questions. 

* When you obtained this log, did you have `50-usb-realtek-net.rules` installed on your system? 
* Also, what would happen if you only installed `51-usb-r8152-net.rules`?

bb-qq, 2025-11-16:
Upon investigation, it appears the “prohibited” message is being output by libsynousb.so, part of Synology's USB system. Since the behavior of this module is undocumented, I cannot determine any further details. Reinstalling DSM may resolve the issue.

Since there has been no response for a long time, this issue is closed. Please feel free to re-open it if you have any questions.