bb-qq/r8152 #541: DS923+ 7.2.2u3 Failed to get IP after months of no issues

powaking, 2025-06-05:
### Description of the problem

NAS suddenly was not accessible over the network. Verified that activity lights were showing traffic yet the NAS was not pingable or reachable over the network. Message seen in /var/log/messages was 'ddnsd.c:2783 Failed to get External IP address'. A reboot of the NAS resolved this issue.

### Description of your products

Synology DS-923+
Linux DS923plus 4.4.302+ #72806 SMP Thu Sep 5 13:41:01 CST 2024 x86_64 GNU/Linux synology_r1000_923+
DSM 7.2.2-72806 Update 3
UGREEN USB to Ethernet Adapter 2.5Gb (RTL8156BG)
Driver package v2.19.2-1 (Around end of January and no issues since)

### Description of your environment

Directly to a Ubiquiti Pro Max 16 switch
Cat5e

### Output of ``dmesg`` command

````
[dmesg.txt](https://github.com/user-attachments/files/20611819/dmesg.txt)
````

### Output of ``lsusb`` command

````
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:06:00.3) hub
  |__1-1         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 65002949B51A2E49)
  |__1-3         0764:0501:0001 00  1.10  1.5MBit/s 50mA 1IF  (CPS ST Series ffffffc7ffffffa2ffffffcbffffffb9)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:06:00.3) hub
  |__2-2         0bda:8156:31f0 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 0F13000000)
````

### Output of ``ifconfig -a`` command

````
bond0     Link encap:Ethernet  HWaddr 90:09:D0:33:9B:59  
          inet addr:192.168.5.237  Bcast:192.168.5.255  Mask:255.255.255.0
          UP BROADCAST MASTER MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

docker0   Link encap:Ethernet  HWaddr 02:42:02:FB:2D:91  
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          inet6 addr: fe80::42:2ff:fefb:2d91/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1308920 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1311186 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:105235752 (100.3 MiB)  TX bytes:479872783 (457.6 MiB)

docker-92 Link encap:Ethernet  HWaddr 02:42:0A:4C:12:43  
          inet addr:172.18.0.1  Bcast:172.18.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

docker1a8 Link encap:Ethernet  HWaddr 36:23:F7:31:D7:D4  
          inet6 addr: fe80::3423:f7ff:fe31:d7d4/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1119966 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1148071 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:76195237 (72.6 MiB)  TX bytes:416104343 (396.8 MiB)

docker33f Link encap:Ethernet  HWaddr 7E:62:DF:E6:BD:D1  
          inet6 addr: fe80::7c62:dfff:fee6:bdd1/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:76400 errors:0 dropped:0 overruns:0 frame:0
          TX packets:69804 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:28733425 (27.4 MiB)  TX bytes:32086584 (30.6 MiB)

docker8e5 Link encap:Ethernet  HWaddr F6:E0:D9:D0:10:7E  
          inet6 addr: fe80::f4e0:d9ff:fed0:107e/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:3 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:379 (379.0 B)  TX bytes:438 (438.0 B)

docker9af Link encap:Ethernet  HWaddr 42:46:3A:64:A7:73  
          inet6 addr: fe80::4046:3aff:fe64:a773/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:12090 errors:0 dropped:0 overruns:0 frame:0
          TX packets:13675 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:1192194 (1.1 MiB)  TX bytes:1968236 (1.8 MiB)

dockerf36 Link encap:Ethernet  HWaddr BA:1F:8E:1F:CD:39  
          inet6 addr: fe80::b81f:8eff:fe1f:cd39/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:98760 errors:0 dropped:0 overruns:0 frame:0
          TX packets:87286 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:17129563 (16.3 MiB)  TX bytes:32396349 (30.8 MiB)

eth0      Link encap:Ethernet  HWaddr 90:09:D0:33:9B:59  
          UP BROADCAST SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:53 base 0x8000  

eth1      Link encap:Ethernet  HWaddr 90:09:D0:33:9B:59  
          UP BROADCAST SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:54 base 0xc000  

eth2      Link encap:Ethernet  HWaddr 6C:1F:F7:1A:04:51  
          inet addr:192.168.5.37  Bcast:192.168.5.255  Mask:255.255.255.0
          inet6 addr: fe80::6e1f:f7ff:fe1a:451/64 Scope:Link
          inet6 addr: 2601:196:8800:24a9:6e1f:f7ff:fe1a:451/64 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:72936730 errors:0 dropped:0 overruns:0 frame:0
          TX packets:121540653 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:33528642302 (31.2 GiB)  TX bytes:129044178297 (120.1 GiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:214119 errors:0 dropped:0 overruns:0 frame:0
          TX packets:214119 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:17061226 (16.2 MiB)  TX bytes:17061226 (16.2 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

````

bb-qq, 2025-06-07:
> https://github.com/user-attachments/files/20611819/dmesg.txt

I could not see this file. Could you re-upload it?

bb-qq, 2025-06-08:
```
2025-06-01T20:14:36-04:00 DS923plus kernel: [9405226.601242] r8152 2-2:1.0: read type=0x01cc, index=0xb40c fail -71
2025-06-01T20:14:37-04:00 DS923plus if_link_down_hook_event[568]: eth2
2025-06-01T20:14:41-04:00 DS923plus ddnsd[9419]: ddnsd.c:2783 Failed to get External IP address
2025-06-01T20:14:41-04:00 DS923plus kernel: [9405232.299184] usb 2-2: device descriptor read/8, error -110
2025-06-01T20:14:45-04:00 DS923plus kernel: [9405235.703786] usb 2-2: device descriptor read/8, error -71
2025-06-01T20:14:53-04:00 DS923plus kernel: [9405243.807114] usb 2-2: device descriptor read/8, error -110
2025-06-01T20:14:58-04:00 DS923plus kernel: [9405248.930569] usb 2-2: device descriptor read/8, error -110
2025-06-01T20:15:06-04:00 DS923plus kernel: [9405257.361242] usb 2-2: device descriptor read/8, error -110
2025-06-01T20:15:07-04:00 DS923plus kernel: [9405257.625400] Module [r8152] is removed. 
2025-06-01T20:15:10-04:00 DS923plus kernel: [9405260.772090] usb 2-2: device descriptor read/8, error -71
2025-06-01T20:15:18-04:00 DS923plus kernel: [9405268.876170] usb 2-2: device descriptor read/8, error -110
2025-06-01T20:15:23-04:00 DS923plus kernel: [9405273.999587] usb 2-2: device descriptor read/8, error -110
2025-06-01T20:15:23-04:00 DS923plus kernel: [9405274.126586] usb usb2-port2: unable to enumerate USB device
```

The logs suggest that the device is completely frozen. 
If this problem occurs under heavy traffic, it could be a heat-related issue, so try placing the device in a well-ventilated area. 
Besides that, I can only suggest trying a different USB port or changing the Ethernet device's vendor to see if that resolves the issue.