bb-qq/r8152 #545: DS415+ slow upload speeds, kernel index fail error

sspeed22, 2025-06-13:
**### Description of the problem**
File copies transfer at less than gigabit speed, Active backup writes only transfer around 27 MB/s when it was transferring at nearly full gigabit speed on internal adapter.

Get a lot of these index fails in the kern.log when copying
[  224.924612] r8152 3-1:1.0: read type=0x0133, index=0xb414 fail -71
[  411.161675] r8152 3-1:1.0: read type=0x01cc, index=0xb400 fail -71
[ 1468.619167] r8152 3-1:1.0: read type=0x0133, index=0xb400 fail -71
[ 1527.756258] perf interrupt took too long (2504 > 5000), lowering kernel.perf_event_max_sample_rate to 50000
[ 1555.760581] r8152 3-1:1.0: read type=0x0133, index=0xb400 fail -71
[ 1574.767477] r8152 3-1:1.0: read type=0x0133, index=0xb5d4 fail -71
[ 1601.803957] r8152 3-1:1.0: read type=0x01cc, index=0xb408 fail -71
[ 1732.981428] r8152 3-1:1.0: read type=0x01cc, index=0xb408 fail -71
[ 1759.016098] r8152 3-1:1.0: read type=0x01cc, index=0xb408 fail -71

**### Description of your products**

Synology DS415+ (avoton) DSM 7.1.1.42962 Update 8
YuLinca 4 Port 2.5G Network Switch with 2x10G SFP
CableCreation 2.5G Ethernet to USB Adapter
r8152-avoton-2.19.2-2_7.1.spk

ESXi 8.0 server -> Unifi 10gbe aggregation switch -> YuLinca -> DS415+

**iperf3 from test client to DS415+**
Server (DS415+) listening on 5201 (test #3)
-----------------------------------------------------------
Accepted connection from 10.1.1.167 (test client), port 58936
[  5] local 10.1.1.1 port 5201 connected to 10.1.1.167 port 58938
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  59.6 MBytes   500 Mbits/sec                  
[  5]   1.00-2.00   sec  77.2 MBytes   647 Mbits/sec                  
[  5]   2.00-3.00   sec  74.5 MBytes   625 Mbits/sec                  
[  5]   3.00-4.00   sec  69.4 MBytes   582 Mbits/sec                  
[  5]   4.00-5.00   sec  72.9 MBytes   612 Mbits/sec                  
[  5]   5.00-6.00   sec  75.3 MBytes   631 Mbits/sec                  
[  5]   6.00-7.00   sec  76.4 MBytes   641 Mbits/sec                  
[  5]   7.00-8.00   sec  73.9 MBytes   620 Mbits/sec                  
[  5]   8.00-9.00   sec  75.8 MBytes   636 Mbits/sec                  
[  5]   9.00-10.00  sec  76.3 MBytes   640 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.00  sec   731 MBytes   613 Mbits/sec                  receiver

**iperf3 from test client to DS1817+ on 10gbe**
Connecting to host 10.1.1.16, port 5201
[  5] local 10.1.1.167 port 41812 connected to 10.1.1.16 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  1.05 GBytes  9.05 Gbits/sec  668    847 KBytes       
[  5]   1.00-2.00   sec  1.09 GBytes  9.35 Gbits/sec  359   1.46 MBytes       
[  5]   2.00-3.00   sec  1.03 GBytes  8.88 Gbits/sec  1155   1.39 MBytes       
[  5]   3.00-4.00   sec  1.08 GBytes  9.25 Gbits/sec  1008   1.37 MBytes       
[  5]   4.00-5.00   sec  1.06 GBytes  9.09 Gbits/sec    3   1.44 MBytes       
[  5]   5.00-6.00   sec  1.05 GBytes  9.05 Gbits/sec  165   1.36 MBytes       
[  5]   6.00-7.00   sec  1.07 GBytes  9.23 Gbits/sec   46   1.40 MBytes       
[  5]   7.00-8.00   sec  1.04 GBytes  8.91 Gbits/sec  308   1.02 MBytes       
[  5]   8.00-9.00   sec  1.04 GBytes  8.90 Gbits/sec   24   1.55 MBytes       
[  5]   9.00-10.00  sec   940 MBytes  7.88 Gbits/sec  228   1.39 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  10.4 GBytes  8.96 Gbits/sec  3964            sender
[  5]   0.00-10.00  sec  10.4 GBytes  8.95 Gbits/sec                  receiver

**iperf3 from DS415+ to test client**
Connecting to host 10.1.1.167 (test client), port 5201
[  5] local 10.1.1.1 port 58263 connected to 10.1.1.167 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   212 MBytes  1.78 Gbits/sec    0    635 KBytes       
[  5]   1.00-2.00   sec   211 MBytes  1.77 Gbits/sec    0    642 KBytes       
[  5]   2.00-3.00   sec   211 MBytes  1.77 Gbits/sec    0    648 KBytes       
[  5]   3.00-4.00   sec   211 MBytes  1.77 Gbits/sec    2    652 KBytes       
[... 122 lines cut ...]

ovs_eth1  Link encap:Ethernet  HWaddr 00:11:32:3A:8A:7A  
          inet addr:169.254.199.75  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

ovs_eth2  Link encap:Ethernet  HWaddr 6C:6E:07:09:09:3A  
          inet addr:10.1.1.1  Bcast:10.1.1.255  Mask:255.255.255.0
          inet6 addr: fe80::6e6e:7ff:fe09:93a/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:694806 errors:0 dropped:0 overruns:0 frame:0
          TX packets:460009 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:1931583764 (1.7 GiB)  TX bytes:4438698334 (4.1 GiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

sspeed22, 2025-06-13:
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.10.108 (root@build3) (gcc version 8.5.0 (GCC) ) #42962 SMP Mon Aug 19 15:12:09 CST 2024
[    0.000000] Command line: root=/dev/md0 ihd_num=4 netif_num=2 HddHotplug=1 pwrctl_pin=N0754 SataPortMap=422 DiskIdxMap=020006 syno_hw_version=DS415+ macs=0011323a8a79,0011323a8a7a,0011323a8a7b,0011323a8a7c sn=14B1MIN431200
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000086fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000087000-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000007ea9efff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ea9f000-0x000000007f3befff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007f3bf000-0x000000007f7befff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007f7bf000-0x000000007f7fefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007f7ff000-0x000000007f7fffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec80000-0x00000000fecfffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed03000-0x00000000fed03fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed08000-0x00000000fed08fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed0c000-0x00000000fed0ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1cfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fef00000-0x00000000feffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffa00000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000027fffffff] usable
[    0.000000] e820: update [mem 0x01d87020-0x01d8702f] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x0000000000086fff] usable
[    0.000000] reserve setup_data: [mem 0x0000000000087000-0x000000000009ffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x0000000001d8701f] usable
[    0.000000] reserve setup_data: [mem 0x0000000001d87020-0x0000000001d8702f] usable
[    0.000000] reserve setup_data: [mem 0x0000000001d87030-0x000000007ea9efff] usable
[    0.000000] reserve setup_data: [mem 0x000000007ea9f000-0x000000007f3befff] reserved
[    0.000000] reserve setup_data: [mem 0x000000007f3bf000-0x000000007f7befff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x000000007f7bf000-0x000000007f7fefff] ACPI data
[    0.000000] reserve setup_data: [mem 0x000000007f7ff000-0x000000007f7fffff] usable
[    0.000000] reserve setup_data: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fec80000-0x00000000fecfffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fed03000-0x00000000fed03fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fed08000-0x00000000fed08fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fed0c000-0x00000000fed0ffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fed1c000-0x00000000fed1cfff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fef00000-0x00000000feffffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000ffa00000-0x00000000ffffffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000100000000-0x000000027fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.40 by INSYDE Corp.
[    0.000000] efi:  SMBIOS=0x7ed06000  ACPI 2.0=0x7f7fe014 
[    0.000000] efi: mem00: type=3, attr=0xf, range=[0x0000000000000000-0x0000000000001000) (0MB)
[    0.000000] efi: mem01: type=7, attr=0xf, range=[0x0000000000001000-0x0000000000085000) (0MB)
[    0.000000] efi: mem02: type=2, attr=0xf, range=[0x0000000000085000-0x0000000000087000) (0MB)
[    0.000000] efi: mem03: type=0, attr=0xf, range=[0x0000000000087000-0x0000000000088000) (0MB)
[    0.000000] efi: mem04: type=6, attr=0x800000000000000f, range=[0x0000000000088000-0x00000000000a0000) (0MB)
[    0.000000] efi: mem05: type=7, attr=0xf, range=[0x0000000000100000-0x00000000768a0000) (1895MB)
[    0.000000] efi: mem06: type=4, attr=0xf, range=[0x00000000768a0000-0x00000000768c0000) (0MB)
[    0.000000] efi: mem07: type=7, attr=0xf, range=[0x00000000768c0000-0x0000000078205000) (25MB)
[    0.000000] efi: mem08: type=2, attr=0xf, range=[0x0000000078205000-0x0000000078bb8000) (9MB)
[    0.000000] efi: mem09: type=4, attr=0xf, range=[0x0000000078bb8000-0x000000007989c000) (12MB)
[    0.000000] efi: mem10: type=2, attr=0xf, range=[0x000000007989c000-0x000000007989f000) (0MB)
[    0.000000] efi: mem11: type=7, attr=0xf, range=[0x000000007989f000-0x0000000079a6a000) (1MB)
[    0.000000] efi: mem12: type=1, attr=0xf, range=[0x0000000079a6a000-0x0000000079a9f000) (0MB)
[    0.000000] efi: mem13: type=7, attr=0xf, range=[0x0000000079a9f000-0x000000007bb4b000) (32MB)
[    0.000000] efi: mem14: type=4, attr=0xf, range=[0x000000007bb4b000-0x000000007bb4c000) (0MB)
[    0.000000] efi: mem15: type=7, attr=0xf, range=[0x000000007bb4c000-0x000000007bb5f000) (0MB)
[... cut ...]
[  212.223360]  [<ffffffff813b7107>] ? sock_ioctl+0x217/0x270
[  212.229691]  [<ffffffff81140237>] ? do_vfs_ioctl+0x797/0x9d0
[  212.236231]  [<ffffffff811295e5>] ? __fput+0x125/0x250
[  212.242073]  [<ffffffff81140513>] ? SyS_ioctl+0xa3/0xb0
[  212.248033]  [<ffffffff8149907e>] ? system_call_fastpath+0x1c/0x21
[  212.255080] ---[ end trace 4cb6a38bf888da7e ]---
[  224.924612] r8152 3-1:1.0: read type=0x0133, index=0xb414 fail -71
[  411.161675] r8152 3-1:1.0: read type=0x01cc, index=0xb400 fail -71

sspeed22, 2025-06-13:
I would like to know about the index fails you see, but I think the problem here is the YuLinca switch.  I get the same crappy performance when I plug the built-in gigabit adapter to the YuLinca switch.

bb-qq, 2025-06-23:
Did you check this page?
https://github.com/bb-qq/r8152/wiki/Troubleshooting