bb-qq/r8152 #554: DS 1821+ with Anker USBC to 2.5Gb ethernet only achieve 1Gb speed

bobottawa, 2025-07-11:
### Description of the problem

Installed r8152-v1000-2.20.1-1_7.2.spk with no problem on DS 1821+ 
Anker USBC to 2.5Gb ethernet only achieves 1Gb speed, not 2.5Gb

### Description of your products

 DS 1821+  DSM 7.22-72806 Update 3
Anker USBC to 2.5Gb ethernet

### Description of your environment

Tried to connect the USBC with a USB-A adaper to the back plane and front USB-A ports.  Both connections are successful, but "Network status" shows 1000 Mbps, Full duplex, MTU 1500
 Ethernet switch: TP-Link 2.5 8 port router, showing a 1Gb/100Mb link light
Cat-6 cable

### Output of ``dmesg`` command

````
(paste output here, or attach as a file)
````

### Output of ``lsusb`` command

````
(paste output here)
````

### Output of ``ifconfig -a`` command

````

Piggy ~> sudo su
Password:
ash-4.4# lsusb
ash: lsusb: command not found
ash-4.4# ifconfig -a
docker0   Link encap:Ethernet  HWaddr 02:42:F8:08:09:23
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          inet6 addr: fe80::42:f8ff:fe08:923/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:438 (438.0 B)

docker-8c Link encap:Ethernet  HWaddr 02:42:EF:4C:F5:4E
          inet addr:172.22.0.1  Bcast:172.22.255.255  Mask:255.255.0.0
          inet6 addr: fe80::42:efff:fe4c:f54e/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:3 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:84 (84.0 B)  TX bytes:438 (438.0 B)

docker-a9 Link encap:Ethernet  HWaddr 02:42:8C:5C:B4:E3
          inet addr:172.27.0.1  Bcast:172.27.255.255  Mask:255.255.0.0
          inet6 addr: fe80::42:8cff:fe5c:b4e3/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:526 errors:0 dropped:0 overruns:0 frame:0
          TX packets:399 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:27304 (26.6 KiB)  TX bytes:39518 (38.5 KiB)

docker-fa Link encap:Ethernet  HWaddr 02:42:5D:F9:72:B5
          inet addr:172.18.0.1  Bcast:172.18.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
[... 165 lines cut ...]
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

tap021132 Link encap:Ethernet  HWaddr 7A:AC:3A:EF:1E:9D
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:2403 errors:0 dropped:0 overruns:0 frame:0
          TX packets:6116 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:311521 (304.2 KiB)  TX bytes:1617762 (1.5 MiB)

tap021132 Link encap:Ethernet  HWaddr AA:04:D1:CE:AF:EE
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:2642 errors:0 dropped:0 overruns:0 frame:0
          TX packets:76204 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:193023 (188.4 KiB)  TX bytes:9188089 (8.7 MiB)

tun0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          inet addr:10.6.0.1  P-t-P:10.6.0.2  Mask:255.255.255.255
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:500
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

bb-qq, 2025-07-12:
Did you check this article?
https://github.com/bb-qq/r8152/wiki/Troubleshooting#reverse-the-orientation-of-the-type-c-connector

bobottawa, 2025-07-12:
I ran  "ethtool eth4" and received the following
Settings for eth4:
        Supported ports: [ MII ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
                                2500baseX/Full
        Supported pause frame use: No
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  10baseT/Half 10baseT/Full
                                             100baseT/Half 100baseT/Full
                                             1000baseT/Full
        Link partner advertised pause frame use: No
        Link partner advertised auto-negotiation: Yes
        Link partner advertised FEC modes: Not reported
        Speed: 1000Mb/s
        Duplex: Full
        Port: MII
        PHYAD: 32
        Transceiver: internal
        Auto-negotiation: on
Cannot get wake-on-lan settings: Operation not permitted
        Current message level: 0x00007fff (32767)
                               drv probe link timer ifdown ifup rx_err tx_err tx_queued intr tx_done rx_status pktdata hw wol
        Link detected: yes

**So the problem is that Advertised link modes doesn't inclue 2500baseT/Full.**
% uname -a 
    Linux Piggy 4.4.302+ #72806 SMP Thu Sep 5 13:45:50 CST 2024 x86_64 GNU/Linux synology_v1000_1821+

%  ethtool -s eth4 autoneg on advertise 0x802f
Cannot set new settings: Operation not permitted
  not setting autoneg

% sudo  ethtool -s eth4 autoneg on advertise 0x802f
return nothing.

**I don't seem to be able to change the Advertised link modes of either eth4 or ovs_eth4.  Any suggestions?**

Thanks so much

bobottawa, 2025-08-02:
Problem solved!  It turns out that the USB-A to USB-C adaptor is a cheap one supporting only USB 2.0 (with a black plastic in the USB A plug).  After switching to a USB3.0 adaptor (with blue plastic in the USB A plug), the connection auto-negotiate 2.5Gb.

Hope this helps others running into the same issue.

closed 2025-08-02 by bobottawa