bb-qq/r8152 #557: Packages does not start after reboot

flightmike, 2025-07-17:
### Description of the problem

> Package does not start after reboot of DSM

### Description of your products

> ds423+
>     - and output of ``uname -a`` command
>     - DSM version 7.2.2-72806 Update 3
> - Ugreen USB 2.5G Lan USB 3.0

### Description of your environment

> -using a ethernet switch 
>     - 
>     - Ethernet switch: Zyxel XMG-108 8 Port
> - cable category Cat.7

### Output of ``dmesg`` command

[\[723408.144033\] ata4 (slot 2) wake.txt](https://github.com/user-attachments/files/21301365/723408.144033.ata4.slot.2.wake.txt)

````

### Output of ``lsusb`` command

|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 6500465045E0FD03)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-2         0bda:8156:3104 00  3.20 5000MBit/s 256mA 2IFs (Realtek USB 10/100/1G/2.5G LAN 4013000001)

### Output of ``ifconfig -a`` command

eth0      Link encap:Ethernet  HWaddr 90:09:D0:6B:01:97
          inet addr:169.254.63.123  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:88 base 0xc000

eth1      Link encap:Ethernet  HWaddr 90:09:D0:6B:01:98
          inet addr:192.168.178.104  Bcast:192.168.178.255  Mask:255.255.255.0
          inet6 addr: fe80::9209:d0ff:fe6b:198/64 Scope:Link
          inet6 addr: fd9e:6d93:b6f4:0:9209:d0ff:fe6b:198/64 Scope:Global
          inet6 addr: 2003:e0:a706:f100:9209:d0ff:fe6b:198/64 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:44650429 errors:0 dropped:0 overruns:0 frame:0
          TX packets:15855800 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:50073922966 (46.6 GiB)  TX bytes:42927674941 (39.9 GiB)
          Interrupt:89 base 0xe000

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:28363 errors:0 dropped:0 overruns:0 frame:0
          TX packets:28363 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:7491595 (7.1 MiB)  TX bytes:7491595 (7.1 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

GJKrupa, 2025-07-20:
I have the same on basically the same hardware (just a different switch - VIMIN 6-Port 2.5G PoE Switch Unmanaged)

bb-qq, 2025-08-31:
Have you run the command in Installation Step 7?
https://github.com/bb-qq/r8152?tab=readme-ov-file#installation

flightmike, 2025-09-04:
Hi, i have run the command in Step 7 and it says, "enabled". But the packages is still stopped and the networkcard is not shown in the List of the DSM.

closed 2025-11-30 by flightmike