bb-qq/r8152 #558: Cannot get this of the aqc111 package to run

iamnaeth, 2025-07-21:
I have a DS1520+.   I have tried to install the package, have it fail.  Run the necessary sudo through putty for both packages and they will not start.  They are stopped and when I select repair it says "failed to run the package service."  I've tried to uninstall the driver / package and repeating.  Neither will start and run.  Not sure what to do from here.

iamnaeth, 2025-07-22:
I was able to install the package by removing all the previous installation.  It is now running without any errors.  However, it does not show up as connected to my router.  Where in the synology would I see the status of the network connection?

TheS1R, 2025-07-22:
In DSM, Network tab within Control Panel.

closed 2025-10-01 by bb-qq