bb-qq/r8152 #563: DS224+ Successful install of driver and adapter working but fails after less than a day

chiggga, 2025-08-03:
### Description of the problem

> I managed to install the correct driver and conenct the adapter to my DS224+, and everything was working well until less than a day later the lights on the adaptor went off. I double checked the adaptor by connecting it to a lan cable and to a TV, and the lights came on, meaning that its not an adaptor issue,

I've uninstalled and reinstalled the driver to no avail. I've tried the front ports too. I've think I've done everything short of getting a new adapter which im averse to. The below outputs are without the adaptor connected, as i cant get access to my NAS with the adaptor. 

One small note, i am able to see my NAS in find.synology.com when using the adaptor, but it shows the IP of LAN 1 instead of LAN 3 or 4.

### Description of your products

> - DS224+
>     - x86_64 GNU/Linux synology_geminilake_224+
>     - DSM version 7.2.2-72806
> - UGREEN UG-CM648-25051 USB TO RJ45 ETHERNET ADAPTER CONVERTER

### Description of your environment

> - NAS connected directly to router via Cat6

### Output of ``dmesg`` command

````

[dmesg.txt](https://github.com/user-attachments/files/21566975/dmesg.txt)

````

### Output of ``lsusb`` command

````
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 65003A692C81C604)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
````

### Output of ``ifconfig -a`` command

````
docker0   Link encap:Ethernet  HWaddr 02:42:37:0A:C7:51
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth0      Link encap:Ethernet  HWaddr 90:09:D0:52:5B:8A
          inet addr:192.168.0.109  Bcast:192.168.0.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:12832 errors:0 dropped:0 overruns:0 frame:0
          TX packets:7926 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:4273119 (4.0 MiB)  TX bytes:7619738 (7.2 MiB)
          Interrupt:84 base 0x2000

eth1      Link encap:Ethernet  HWaddr 90:09:D0:52:5B:8B
          inet addr:169.254.241.88  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:85 base 0x4000

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:18401 errors:0 dropped:0 overruns:0 frame:0
          TX packets:18401 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:9759257 (9.3 MiB)  TX bytes:9759257 (9.3 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

bb-qq, 2025-08-31:
> [dmesg.txt](https://github.com/user-attachments/files/21566975/dmesg.txt)

I could not see this log.

bb-qq, 2026-08-10:
Closing due to no follow-up. Please reopen with the lsusb output if the device still isn't detected at the USB level.