bb-qq/r8152 #566: DS720+ with UGreen 25051 USB-A to RJ45 2.5G Ethernet Adapter

jonhp, 2025-08-16:
Followed the driver installation instructions on the r8152 github home page completely.
DSM gui shows "Error" in red.
synosystemctl enable pkgctl-r8152: Fail to enable [pkgctl-r8152]
Driver version: 2.20.1-1
DSM version: 7.2.2-72806

I thought I had read that someone had this combination working, but I guess not.

PS: The release note installation instruction does not exactly match that given on the r8152 github home page. For a first timer this is confusing. I would suggest that your release notes refer back to the r8152 github home page instructions which I think are more thorough.

bb-qq, 2025-08-31:
Could you attach these logs?
https://github.com/bb-qq/r8152/wiki/Troubleshooting#how-to-investigate-if-the-driver-does-not-work-properly

jonhp, 2025-08-31:
can't reproduce.

closed 2025-08-31 by jonhp